MillenniumMachines/MillenniumOS #72: Wide radius bits aren't raised to safe levels before attempting to probe outside rad

bdistin, 2024-05-03:
[image]

benagricola, 2024-05-07:
As discussed in Discord, this is down to the dive height configured on the toolsetter - after probing the centre of a tool with a radius wider than the toolsetter surface, the tool will be backed off by the dive height before attempting to probe at the tool radius - in this case the dive height of 5mm was too low, causing MillenniumOS to attempt to move one of the inserts through the toolsetter surface.

We'll need to add a note to this (or a safety check during toolsetting) to make sure that the operator has their dive height set appropriately.

benagricola, 2024-06-12:
There's now a warning in tutorial mode to check for the clearance based on the dive height set on the probe.

closed 2024-06-12 by benagricola