MillenniumMachines/MillenniumOS #141: Spindle starts after cancelling the job

vvviiimmm, 2024-09-16:
1. start any job, probe/set tools if needed
2. click "Pause" when "CAUTION: Spindle #0 will now start!" pop up is shown
3. click "Cancel Job" in "Job control" section
 
Observed behavior: **spindle starts**
Expected behavior: spindle does not start, the job is safely cancelled

Version: v0.3.1

<details>
<summary>all variables</summary>

M98 P"0:/macros/MillenniumOS/4. Misc/Print All Variables.g"
=== MOS Info:
      global.mosVer=v0.3.1
      global.mosErr=null
      global.mosLdd=true
=== MOS Features:
      global.mosFeatToolSetter=true
      global.mosFeatTouchProbe=false
      global.mosFeatSpindleFeedback=false
      global.mosFeatVSSC=true
=== MOS Probing:
      global.mosPTID=49
      global.mosPD=null
      global.mosDPID=null
      global.mosPCX=318.0000
      global.mosPCY=141.0000
      global.mosPCZ=-83.4530029
      global.mosPVX=0.0000000
      global.mosPVY=0.0000000
      global.mosPVZ=0.0000000
      global.mosOT=2.0
      global.mosCL=10.0
      global.mosWPCtrPos={{null,null},{null,null},{null,null},{null,null},{null,null},{null,null},{null,null},{null,null},{null,null}}
      global.mosWPDims={{null,null},{null,null},{null,null},{null,null},{null,null},{null,null},{null,null},{null,null},{null,null}}
      global.mosWPRad={null,null,null,null,null,null,null,null,null}
      global.mosWPDeg={null,null,null,null,null,null,null,null,null}
      global.mosWPCnrNum={null,null,null,null,null,null,null,null,null}
      global.mosWPCnrPos={{null,null},{null,null},{null,null},{null,null},{null,null},{null,null},{null,null},{null,null},{null,null}}
      global.mosWPCnrDeg={null,null,null,null,null,null,null,null,null}
      global.mosWPSfcAxis={null,null,null,null,null,null,null,null,null}
      global.mosWPSfcPos={null,null,null,null,null,null,null,null,null}
=== MOS Touch Probe:
      global.mosTPID=null
      global.mosTPR=null
      global.mosTPD=null
      global.mosTPRP=null
=== MOS Toolsetter:
      global.mosTSID=1
      global.mosTSP={318.000,141.000,-69.9110031}
      global.mosTSAP=null
=== MOS Misc:
      global.mosPMBO=0.5
=== MOS Spindle:
      global.mosSID=0
      global.mosSAS=1
      global.mosSDS=1
</details>

vvviiimmm, 2024-09-16:
(sorry for reporting instead of fixing, don't have the time atm)

benagricola, 2024-09-16:
> (sorry for reporting instead of fixing, don't have the time atm)

All good, thanks for reporting this.

The MillenniumOS code that triggers this _should not_ be executed by RepRapFirmware once the cancel button is pressed, but it looks like RRF is continuing to process the spindle startup macro even when the cancel button is clicked. It's been reported to one of the RRF devs to investigate. Will update here when I know more.

benagricola, 2024-09-17:
Update: It's unlikely that the underlying behaviour in RRF will be fixed in short order (this is down to how macros are paused / restarted in RRF). In this case, it is a safety issue since we should never be starting the spindle if the user did not click start on the confirmation box.

Given that, I have removed the "pause" option from the confirmation box since the behaviour when it is then cancelled via the UI is undefined.

benagricola, 2024-10-03:
This behaviour has been merged into main and is in the pre-releases of v0.4.0.

closed 2024-10-03 by benagricola